lancaster-university/microbit-v2-samples #9: Missing BLE Sample

microbit-sam, 2020-11-18:
microbit-v2-samples doesn't have an example for BLE
Should Martin's BLE test program be pulled in?

https://github.com/martinwork/codal/blob/ble2-work/source/BLETest.needsble

adamstark, 2021-01-15:
I'd also love to see an example for BLE - is one available? :)

microbit-sam, 2021-01-19:
Hey Adam, I've created a PR with the examples from micro:bit v1 running on v2 that may interest you
https://github.com/lancaster-university/microbit-v2-samples/pull/19

adamstark, 2021-01-19:
Thanks @microbit-sam - I actually did some hacking around and got something working myself but this looks much more complete than what I've managed. Thank you! :)

RoufaidaLaidi, 2022-04-03:
Hello, 
I would like to make a code to allow two Microbits to communicate over BLE (central and peripheral), is this possible ? And if so, how to start ?

microbit-carlos, 2023-01-03:
I'll close this issue as the PR with the BLE samples has been merged.

@RoufaidaLaidi To have one of the micro:bits configured as BLE central the CODAL project needs to be configured with SoftDevice S140. It currently uses S113, which only contains peripheral mode, as that saves some flash and RAM.
You can update CODAL to use s140 instead, which it did originally a couple of years ago. The best way to start would be to have a look at this PR, and the PRs listed in the description, and revert them on top of the latest release:
- https://github.com/lancaster-university/codal-microbit-v2/pull/3

If you have any question with respect to that please open an issue in the `codal-microbit-v2` repo and we can help you over there: - https://github.com/lancaster-university/codal-microbit-v2/issues

closed 2023-01-03 by microbit-carlos