lancaster-university/microbit-v2-samples #13: speaker_test() sample no longer works

dpgeorge, 2020-12-02:
Since the changes to the audio mixer pipeline, the provide `speaker_test()` sample no longer works.

finneyj, 2020-12-02:
Thanks @dpgeorge 

Yes, I haven't had time to update all the samples - sorry  :(

Suggest we deprecate this in favour of the say_hello() function just above it:
https://github.com/lancaster-university/microbit-v2-samples/blob/master/source/samples/SpeakerTest.cpp#L926

Which provide equivalent functionality through the mixer pipeline.

Does that provide what you were looking for?

dpgeorge, 2020-12-02:
I actually wanted to use the raw NRF52PWM object, to bypass the mixer and do custom DAC-like output.  I don't have a need for this anymore, but that was my reason to try and use `speaker_test()`.

finneyj, 2020-12-02:
ah, I see - thanks.

Most likely this is simply because the NRF_PWM module being used in the sample is already in use.... When I get a moment I'll flip it to an unused module and test it works ok.