lancaster-university/microbit-v2-samples #14: Building on windows: issue and solution.

LovelyXandy, 2020-12-02:
Took me a while to work out - the build.py (and associated scripts) on a windows machine fail with a not very useful error if the directory path contains a space character (which github desktop defaults to including); they also won't download the libraries from the other githubs unless the path to the git command line executable is added (which the desktop github for windows doesn't do automatically).

finneyj, 2020-12-02:
Thanks for reporting @LovelyXandy. Sorry to hear you didn't have the smoothest ride here.

Perhaps you could advise a fix when you have a bit of spare time @jamesadevine? I think you know that script better than anyone?

jamesadevine, 2020-12-02:
Have you got a repro @LovelyXandy?

martinwork, 2020-12-10:
I think https://github.com/lancaster-university/microbit-v2-samples/pull/10 will help build.py work with spaces.

I haven't hit the problem with the git command path.

microbit-carlos, 2023-01-19:
PR created:
- https://github.com/lancaster-university/codal/pull/50

closed 2023-01-19 by microbit-carlos