lancaster-university/microbit-v2-samples #21: Feature request: Add battery level indicator sample

microbit-mark, 2021-01-26:
@jaustin created this example for V1 that shows the current battery output as Volts on the display. 
https://os.mbed.com/teams/microbit/code/microbit-battery-test/

We surface this in 
https://support.microbit.org/solution/articles/19000069531-measuring-battery-capacity

Could we have a similar example using CODAL?

njheb, 2022-03-19:
[hacked-together-sample-vdd-on-startup.zip](https://github.com/lancaster-university/microbit-v2-samples/files/8309418/hacked-together-sample-vdd-on-startup.zip)

Maybe this will do as the basis. If I have time I'll neaten up the assembler.

njheb, 2022-03-20:
[uBit-v2-sample-VDD-on-boot.zip](https://github.com/lancaster-university/microbit-v2-samples/files/8311201/uBit-v2-sample-VDD-on-boot.zip)
Might this do as the basis for an example. 
Tested only on my only uBit:v2. 
I guessed at typical diode forward drop value.
HEX file included in zip along with source.