lancaster-university/microbit-v2-samples #25: Random disconect after ble connection

semakers, 2021-02-24:
After connecting with the example of ble on my terminal luetooth on android the device (microbitv2) disconnects after a random amount of time.

Ziriax, 2021-04-30:
I'm also experiencing bluetooth "crashes" with v2. 

I have a micro:bit v1, and I connect with web bluetooth services, plotting all sensor values over time. The v1 keeps running for hours.

However, when using the same code with the v2, it "disconnects" after a while (actually, it seems to crash, because the disconnect event handler is not called on the device anymore).

I have now enabled 

```
 "MICROBIT_DMESG_LEVEL": 4,
 "DMESG_SERIAL_DEBUG": 1
```

in the codal config file, and I'm monitoring all serial debug messages, in the hope I can see what is going wrong.

Ziriax, 2021-04-30:
I'm not sure if it helps, but this is the serial log before the bluetooth services stopped working (note that I'm not seeing a disconnection, just no more data is being sent to the web browser).

I will make the source code with Dockerfiles etc available, so maybe someone can try to reproduce this on a different micro:bit and different OS (I'm using Chrome on a Windows 10 PC). I'll try myself on Chrome/MacOS.

```
Graph Event: :notifyChrValue 38 (for calling sd_ble_gatts_hvx( 68, 0x20006224,roBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 38,622A, 53, 2)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_bx20005F6C, 176, 6)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 71, 0x2000622A, 55, 2)
Graph Event: :nr::notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224172, 6)
Graph Event: :notifyChrValue 38 (for MicroBitBLEChar)
Console Data: calling sd_ble_ga0x20006224, 170, 6)
Graph Event: :notifyChrValue 71 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 71, 0x2000622A, 54, 2)
Graph Event: :_ble_gatts_hvx( 71, 0x2000622A, 53, 2) (for MicroBitBLEChar)
Graph Event: :notifing sd_ble_gatts_hvx( 68, 0x20006224, 232, 6) (for MicroBitBLEChar)
Graph Event: :notifyChrValue 68 (for MicroBitBLECharChar)
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006, 200, 6)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_ble_ling sd_ble_gatts_hvx( 80, 0x20006268, 31, 1)
Console Data: MicroBitBLECharlling sd_ble_gatts_hvx( 68, 0x20006224, 38, 6)
Console Data: MicroBitBLECha38
Console Data: calling sd_ble_gatts_hvx( 38, 0x20005F6C, 168, 6)
Console Data: MicroBialue 68
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224, 82, 6)
Console Data: Miclue 68
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224, 250, 6)
Graph Event: :notifyChrValue 38 (for MicEChar)
Graph Event: :notifyChrValue 71 (for calling sd_ble_gatts_hvx( 38, 0x2000tBLEChar)
Console Data: calling sd_ble_gatts_hvx( 71, 0x2atts_hvx( 68, 0x20006224, 82, 6)
Graph Event: :notifyChrVable_gatts_hvx( 38, 0x20005F6C, 168, 6) (for MicroBitBLEChar)
Graph Event: :notif (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 38, 0x20005F6C, 172, 6)
Console Data: MicroBitBLifyChrValue 71
Graph Event: :notifyChrValue 68 (for calling sd_ble_gatts_hvx( 71, 0x2000622A, 53, r)
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224vx( 38, 0x20005F6C, 176, 6)
Graph Event: :notifyChrValue 6gatts_hvx( 38, 0x20005F6C, 180, 6) (for MicroBitBLEChar)
Graph Event: :notifyChrtts_hvx( 71, 0x2000622A, 55, 2) (for MicroBitBLEChar)
Graph Event: :notifyChrVallue 68 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224, 232, 6)
Console Data: Mic_ble_gatts_hvx( 38, 0x20005F6C, 172, 6)
Graph Event: :notitBLEChar::notifyChrValue 38 (for MicroBitBLEChar)
Graph Event: :notifyChrValue 38 (for calling sd_ble_gatts_hvx( 38, 0x2BitBLEChar)
Graph Event: :notifyChrValue 71 (for calling sd_ble_gatts_hvx( 38, 0croBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 71alue 68
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224, 250, 6)
Graph Event: :notifyChrValue 38 (for MiBLEChar)
Console Data: calling sd_ble_gatts_hvx( 38, 0x20WON FRAME
Graph Event: :notifyChrValue 38 (for MicroBitBLEChar)
Graph Event: :notifyChrValue 80 (for calling sd_ble_MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 0, 6)
Graph Event: :notifyChrValue 71 (for MicroBitBLEChar)
Console Data: calling sd_ble_gattValue 68
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224, 2, 6)
Console Data: MictifyChrValue 38
Graph Event: RECEIVED UNKNWON FRAME (for calling sd_ble_gatts_hvx( 38, 0x20005F6C, 180UIPM)
Graph Event: :notifyChrValue hvx( 71, 0x2000622A, 53, 2) (for MicroBitBLEChar)
Graph Event: :notifyChrValue 320006224, 144, 6) (for MicroBitBLEChar)
Graph Event: :notifyChrValue 71 (for MicroBitBLEChar)
Console Data: calling5F6C, 180, 6)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_x2000622A, 55, 2)
Graph Event: :notifyChrValue 80 (for MicroBitBLEChar)
Console Data: calling)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hv, 6)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
[... 213 lines cut ...]
Console Data: calling s53, 2)
Graph Event: RECEIVED UNKNWON FRAME (for UIPM)
Graph Event: :notifyCalling sd_ble_gatts_hvx( 68, 0x20006224, 188, 6) (for MicroBitBLEChar)
Console Data: MicroBitBLEC180, 6)
Graph Event: :notifyChrValue 38 (for MicroBitBLEChar)
Console Data: calling sd_ble_ga4, 38, 6)
Graph Event: :notifyChrValue 71 (for MicroBitBLEChar)
Console Data: calling sd_ble_0x20005F6C, 160, 6)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calli0x20006224, 232, 6)
Graph Event: :notifyChrValue 71 (for MicroBitBLEChar)
Console Data: calli_gatts_hvx( 71, 0x2000622A, 53, 2)
Graph Event: :notifyChralue 38 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 38, 0x20005F6C, 180, 6)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224, 144, 6)
Graph Event: :notifyChrValue 71 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 71, 0x2000622A, 53, 2)
Graph Event: :notifyChrValue 68 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 68, 0x20006224, 144, 6)
Graph Event: :notifyChrValue 71 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 71, 0x2000622A, 53, 2)
Graph Event: :notifyChrValue 80 (for MicroBitBLEChar)
Console Data: calling sd_ble_gatts_hvx( 80, 0x20006268, 31, 1)
```

Ziriax, 2021-04-30:
Oops, I forgot we need to raise issues in https://github.com/lancaster-university/codal-microbit-v2...