lancaster-university/microbit-v2-samples #28: GCC ARM not working on macOS Catalina despite being installed and added to PATH

tmiliant, 2021-03-11:
The following is the error I got from the terminal after running "python build.py":

codal-microbit-v2 is already installed
Set target: codal-microbit-v2 
Using target.json (dev version) 
Targeting codal-microbit-v2
Supressing -Wexpansion-to-defined.
-- Adding library path: (/Users/valentine_qkd/Desktop/microbit-v2-samples-master/libraries)
Installing dependencies...
codal-core is already installed
codal-nrf52 is already installed
codal-microbit-nrf5sdk is already installed
Using library: codal-microbit-v2
Using library: codal-core
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Using library: codal-nrf52
Using library: codal-microbit-nrf5sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/valentine_qkd/Desktop/microbit-v2-samples-master/build
[  1%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/core/CodalComponent.cpp.o
[  1%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/core/CodalFiber.cpp.o
[  1%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/core/codal_default_target_hal.cpp.o
[  1%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/core/CodalHeapAllocator.cpp.o
[  1%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/driver-models/Timer.cpp.o
[  2%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/driver-models/AbstractButton.cpp.o
[  2%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/driver-models/Sensor.cpp.o
[  2%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/drivers/HIDMouse.cpp.o
[  2%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/drivers/Button.cpp.o
[  3%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/drivers/ILI9341.cpp.o
clang: clangwarning: : warning-Wl,--gc-sections: 'linker' input unused [-Wunused-command-line-argument]: 
-Wl,--gc-sections: 'linker' input unused [-Wunused-command-line-argument]clang: 
warningclang: : -Wl,--sort-common: 'linker' input unused [-Wunused-command-line-argument]warning: 
-Wl,--sort-common: 'linker' input unused [-Wunused-command-line-argument]clang: 
warningclang: : -Wl,--sort-section=alignment: 'linker' input unused [-Wunused-command-line-argument]warning: 
-Wl,--sort-section=alignment: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mcpu=cortex-m4' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mthumb' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=fpv4-sp-d16' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=softfp' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mcpu=cortex-m4' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mthumb' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=fpv4-sp-d16' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=softfp' [-Wunused-command-line-argument]
clang: warning: -Wl,--gc-sections: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-common: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-section=alignment: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mcpu=cortex-m4' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mthumb' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=fpv4-sp-d16' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=softfp' [-Wunused-command-line-argument]
clang: warning: -Wl,--gc-sections: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-common: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-section=alignment: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mcpu=cortex-m4' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mthumb' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=fpv4-sp-d16' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=softfp' [-Wunused-command-line-argument]
clang: warning: -Wl,--gc-sections: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-common: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-section=alignment: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mcpu=cortex-m4' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mthumb' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=fpv4-sp-d16' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=softfp' [-Wunused-command-line-argument]
clang: warning: -Wl,--gc-sections: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-common: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: -Wl,--sort-section=alignment: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mcpu=cortex-m4' [-Wunused-command-line-argument]
[... 254 lines cut ...]
1 warning generated.
make[1]: *** [libraries/codal-core/CMakeFiles/codal-core.dir/all] Error 2
make: *** [all] Error 2

I installed the tar file from the photo below(should I have installed the package as well?):

[image]

Then I expanded the tar file and moved the result to /usr/local with "sudo mv (download path) /usr/local/gcc-arm".

The following are my paths in /etc/paths, after adding the binaries to the path:
/usr/local/bin
/usr/bin
/bin
/usr/sbin
/sbin
/usr/local/gcc-arm/bin

The following is the path of interest in ~/.bash_profile:
export PATH=$PATH:/usr/local/gcc-arm/bin

I did all of the above on two different macOS, both with Catalina on them. Same errors in the same order.

Below is photo showing I have everything that is required installed:
[image]

jamesadevine, 2021-03-11:
I'd check to see if the terminal window you are compiling from is using the updated path.

Would be great to see your cmake version if possible...

tmiliant, 2021-03-11:
I had also installed arm-none-eabi-gcc using "brew install ..." that is why at the end of the issue it shows up with 2019 instead of 2020. When I try to invoke arm-none-eabi-gcc as suggested by the readme.txt, it gives me this:
[image]

tmiliant, 2021-03-11:
I also tried running "python build.py" without having installed anything related to gcc arm. The exact same errors appear. This means that the process above is not doing anything, although I seem to have followed everything step by step.

tmiliant, 2021-03-11:
[image]

tmiliant, 2021-03-11:
[image]
[image]

tmiliant, 2021-03-11:
[image]

I installed cmake using "brew install cmake"

tmiliant, 2021-03-11:
Where should the build.py file be located?

tmiliant, 2021-03-11:
What is the logic behind it? How do I know which compiler the command "python build.py" is using?

When running "python build.py", it's fine because "which python" gives "/usr/bin/python" and "/usr/bin" is in the PATH. 

However the command "python build.py" does not contain "arm-none-eabi-gcc" which would make sense as "which arm-none-eabi-gcc" gives "/usr/local/gcc-arm/bin/arm-none-eabi-gcc" and I added "/usr/local/gcc-arm/bin/" to the path. 

By this logic should it not be "arm-none-eabi-gcc build.py"? Which part of "python build.py" is invoking the binaries of gcc-arm?

closed 2021-03-14 by tmiliant