lancaster-university/microbit-v2-samples #29: NRF52833

tcpipchip, 2021-03-14:
I finally compiled using Msys32
[image]

tcpipchip, 2021-03-14:
the HEX created is a hello world, right ?
How can i compile the other examples ?

finneyj, 2021-03-14:
Oh, that's cool. Perhaps you could share the process so that other can more easily do the same? We have a docs site coming soon, and I'd welcome a PR to the README?

You'll find quite a few examples in the "samples" folder of this repo. I'm afraid I haven't had chance to organize the samples here as well as I did for micro:bit v1 (Universities have been really busy due to all the covid mitigation work we've had to do this last 12 months... so there hasn't been much time for voluntary projects!). please consider these a work-in-progress. Most samples are fine, but I expect a few may be a little out of date. Please raise an issue here if you have problems.

The process to run a samples is simple though. They are all compiled by default, you just have to call the relevant function from main.cpp (in the source folder):

- #include "tests.h"
- delete the hello world code
- call the relevant samples function from the samples folder instead.
- run "python build.py" to generate the new HEX file

I hope this helps

tcpipchip, 2021-03-14:
i am so sorry
I uninstall msys2 and start the install step to step...and you dont believe, not work more ;(

Install MSYS2 (Windows)
https://www.msys2.org/
[image]
[image]
[image]
Open MSYS2
[image]
Install GCC
[image]
[image]
install GIT
[image]
[image]
install CMAKE
[image]
[image]
install make
[image]
install gcc-arm-none-eabi
[image]
[image]
install binutils-arm-none-eabi
[image]
[image]
install python3
[image]
[image]
clone the Microbit V2 github
git clone https://github.com/lancaster-university/microbit-v2-samples.git
[image]
enter in microbit-v2-samples
[image]
compile
[image]
[image]

tcpipchip, 2021-03-14:
during compile python3 build.py got strange erros :(
So sorry!

tcpipchip, 2021-03-15:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
If got error, remove --enable-auto-import and 1 from W on cmakelist

tcpipchip, 2021-03-15:
when you are back, let me know
i have questions!

finneyj, 2021-03-15:
Thanks very much for this @tcpipchip!

It will be very useful to be able to point others at your resource here, so that they can benefit from what you've learned.

If you have questions, feel free to raise them in an issue. We'll try to answer them as soon as we can. :)

tcpipchip, 2021-03-16:
Thanks @finneyj 

About the .HEX/BIN created by the MicroBit V2 compiler,  includes de BOOTLOADER of the NORDIC NRF52833 ?

For example, if the user not use the MASS STORAGE DEVICE (DAP-LINK) to transfer, but just a SWD (SEGGER, ST-LINKV V2) to transfer, will work ?

basicchip, 2021-03-16:
I'm following in tcpipchip's footsteps running on MSys.  I can compile and build files in microbit-ve-samples, but when I copy the resulting .hex file to MAINTENANCE   it shows a FAIL text document

error: The starting address for the interface update is wrong.
type: user

This is with the default hello world example  --   Any clues?

Has anyone a known good .hex file for a microbit v2

jamesadevine, 2021-03-16:
You should not be copying the resulting hex file to maintenance. Copy the file to the standard MICROBIT mass storage device.

finneyj, 2021-03-16:
@tcpipchip The MICROBIT.HEX file produced is a full image for the NRF52833 MCU. 

Using the DAPLink interface is the simplest way to get that image onto the NRF52833, but if you prefer to use a different SWD programmer that should work fine too.

finneyj, 2021-03-16:
@basicchip Does your micro:bit always boot into MAINTENANCE mode? It really shouldn't...

If it does, this implies that the KL27 interface MCU has somehow lost its DAPLINK firmware. You can recover by reflashing the DAPLINK firmware by drag/drop of the latest DAPLInk firmware in whilst the device is in MAINTENENCE mode. it should then reboot into normal MICROBIT mode after that.

You can get the DAPLink firmware from the micro:bit foundation website. Make sure you use the V2 version :)

https://tech.microbit.org/software/daplink-interface/

basicchip, 2021-03-16:
Thank you finneyj, sorry I misread the download procedure.  Hello World now comes up on my microbit V2

Can you use the DAPLINK to do any GDB debugging, with pyocd for instance?

jamesadevine, 2021-03-16:
@basicchip Yes absolutely.

Here is the pyocd command I use:
```
pyocd-gdbserver --persist -t nrf52840 -bh -r
```
You must use the nrf52840 target for pyocd to get a functional memory map (nrf52833 is not yet added).

In the micro:bit samples v2 directory: 
```
arm-none-eabi-gdb --eval "target remote :3333" build/MICROBIT 
```

basicchip, 2021-03-16:
Looking for a UART example (there is one in the codal v2 samples), not sure I can mix and match.  I have not got docker running on Msys, but I believe I got it running on debian, but need to switch PCs

tcpipchip, 2021-03-16:
@jamesadevine 
@finneyj 
I have here in stock some NRF52832 and NRF52840 modules.
Is it possible modile the compiler to port the code created to them ?

jamesadevine, 2022-07-12:
@tcpipchip this bootloader exists from adafruit: https://github.com/adafruit/Adafruit_nRF52_Bootloader I believe I added support for the 833 some time ago.

jamesadevine, 2022-07-12:
I'm afraid not -- I no longer have a board with the bootloader installed. You'll have to try it yourself!

tcpipchip, 2022-07-12:
yea, i can do...when back home! Thanks!

tcpipchip, 2022-07-12:
Wow, the python converter fail to convert the hex to dfu, to me transfer to the nrf52833 :(

tcpipchip, 2022-07-13:
Success! Running Microbit no U-BLOX NINA B4 :)
I had to edit the HEX, because is to in one!

tcpipchip, 2022-07-20:
https://nina-b302-arduino-adafruit.blogspot.com/2022/07/u-blox-nina-b40-opencpu-programado-com.html