lancaster-university/microbit-v2-samples #33: Does not compile with GCC 10.3

microbit-carlos, 2021-09-07:
Released last July.

Right now the CI in this repo fails on the macOS build, this is because brew tap we use has updated the arm gcc formula with v10.3:
- brew formula: https://github.com/ARMmbed/homebrew-formulae/blob/master/arm-none-eabi-gcc.rb
- CI failure: https://github.com/lancaster-university/microbit-v2-samples/runs/3531365675?check_suite_focus=true

```
[ 18%] Building CXX object libraries/codal-core/CMakeFiles/codal-core.dir/source/drivers/MAG3110.cpp.o
/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T//ccCCqnhY.s: Assembler messages:
/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T//ccCCqnhY.s:124: Warning: ignoring changed section attributes for .data
/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T//ccCCqnhY.s:136: Warning: dwarf line number information for .data ignored
/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T//ccCCqnhY.s:141: Warning: dwarf line number information for .data ignored
/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T//ccCCqnhY.s:151: Warning: dwarf line number information for .data ignored
/var/folders/24/8k48jl6d249_n_qfxwsl6xvm0000gn/T//ccCCqnhY.s:15751: Error: leb128 operand is an undefined symbol: .LVU24
make[2]: *** [libraries/codal-core/CMakeFiles/codal-core.dir/source/driver-models/Timer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [libraries/codal-core/CMakeFiles/codal-core.dir/all] Error 2
make: *** [all] Error 2
Error: Process completed with exit code 1.
```

I've tested this in my fork testing in Ubuntu with version `10.2.1` (`10-2020-q4`) and that works, then on version `10.3.1` (`10.3-2021.07`) and it throws the same error.
- 10.2.1: https://github.com/microbit-carlos/microbit-v2-samples/runs/3525637382?check_suite_focus=true
- 10.3.1: https://github.com/microbit-carlos/microbit-v2-samples/runs/3532803826?check_suite_focus=true

robyoung, 2021-11-21:
can confirm that it does compile with the `10-2020-q4-major` release from the [GNU Arm Embedded Toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads#panel3a) from the Arm Developer site.

finneyj, 2022-01-13:
Perfect - thanks John. Great to have you on board!

microbit-carlos, 2022-01-13:
Does that mean those functions will be copied to RAM to be executed from there? I assume that will increase the memory footprint?

JohnVidler, 2022-01-13:
The fallback op-counting timer was already running in RAM to begin with, although it is extremely small.

This change would still do this, but simply place it in the correct ELF section to make the linker happy, it'll still be in the final `.data` block either way.

JohnVidler, 2022-01-13:
Tested the patch on an rp2040 and a Micro:bit, both seem happy, so tentatively placed the patch in a new branch over on the `codal-core` repo; https://github.com/lancaster-university/codal-core/tree/compiler/gcc-10.3.x

If you `git checkout` the `compiler/gcc-10.3.x` branch in your local `libraries/codal-core` path, it should now compile with gcc 10.3.x +

JohnVidler, 2022-01-13:
Pull request created over in `codal-core` to merge these into master, if folks are happy its not breaking anything I might have missed: https://github.com/lancaster-university/codal-core/pull/144

JohnVidler, 2022-01-14:
Merged into master, (see https://github.com/lancaster-university/codal-core/pull/144 ) so this should all be fixed 👍 

Closing this for now, but feel free to reopen if anyone finds issues.

microbit-carlos, 2022-01-14:
Awesome, thanks @JohnVidler!

closed 2022-01-14 by microbit-carlos