lancaster-university/microbit-v2-samples #34: Installation and Building

TDanielI, 2021-09-29:
Hello
I have a question about the building section of the GitHub:
Building

Clone this repository
In the root of this repository, type python build.py
The hex file will be built MICROBIT.HEX and placed in the root folder.

What does point 2 mean, should I type it on the python console?
Also, just to confirm my understanding about the developing section, is it true that I can compile any file with .h .c and .cpp format by placing the files on the source file?
Thank you!

carlosperate, 2021-09-29:
> In the root of this repository, type python build.py
>...
>What does point 2 mean, should I type it on the python console?

It means that, in the command prompt, make sure you are in the folder where you clones this repository, and then you can run `python build.py` (or on Windows likely `py build.py`) to run the `build.py` script.

> Also, just to confirm my understanding about the developing section, is it true that I can compile any file with .h .c and .cpp format by placing the files on the source file?

Yeah, you can drop all your `.h` and `.c` files in the `source` directory.

carlosperate, 2021-09-29:
Is GCC Arm set up in your path?
If you execute this on the same terminal, do you get the expected gcc version?

```
arm-none-eabi-gcc --version
```

TDanielI, 2021-09-29:
I checked on the environment variables and found that the path was only installed locally, i duplicated it to the system and it failed with another long error, it failed after [1/215] Building CXX object libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/PacketBuffer.cpp.obj

carlosperate, 2021-09-29:
Could you copy the entire output of the build command? That's not quite enough to see the error.

TDanielI, 2021-09-29:
FAILED: libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/PacketBuffer.cpp.objC:\PROGRA~2\GNUARM~1\102020~1\bin\AR10B2~1.EXE  -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/inc" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/inc/bluetooth" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/inc/compat" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/model" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/inc" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/inc/cmsis" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/mdk" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/templates" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/templates/nRF52833" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/drivers/include" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/drivers/src" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/core" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/driver-models" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/drivers" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/streams"-I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/types" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/../codal-nrf52/nrfx/hal" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/sdk_config" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK_mods" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/pwr_mgmt"-I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/sortlist" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/strerror" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/softdevice/common" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/crc32" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/bootloader/dfu" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/util" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/common" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/balloc" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/peer_manager" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/ringbuf" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/timer" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/log" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/nrf_ble_gatt" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/bootloader" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/fstorage" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/experimental_section_vars" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/softdevice/s113/headers" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/mutex" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/delay" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/bootloader/ble_dfu" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/atomic_fifo" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/atomic" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/memobj" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/fds" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/atomic_flags" -I"C:/Users/T Daniel/Pyc [... cut ...] MF DEPFILE -o libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/NRF52LedMatrix.cpp.obj -c "C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/source/NRF52LedMatrix.cpp"AR10B2~1.EXE: error: Daniel/PycharmProjects/microbit-v2-samples/build/codal_extra_definitions.h: No such file or directoryAR10B2~1.EXE: error: Daniel/PycharmProjects/microbit-v2-samples/utils/cmake/toolchains/ARM_GCC: No such file or directoryninja: build stopped: subcommand failed.

carlosperate, 2021-09-29:
It's really hard to read the output if it's all in a single line. Could you update your comment using triple back quotes?

Like in this screenshot:
[image]

TDanielI, 2021-09-29:
There are many long lines without new lines, but I tried to clean that up, I hope this helps:
```FAILED: libraries/codal-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/PacketBuffer.cpp.obj
C:\PROGRA~2\GNUARM~1\102020~1\bin\AR10B2~1.EXE  -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/inc" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/inc/bluetooth" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/inc/compat" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/model" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/inc" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/inc/cmsis" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/mdk" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/templates" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/templates/nRF52833" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/drivers/include" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-nrf52/nrfx/drivers/src" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/core" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/driver-models" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/drivers" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/streams"-I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-core/./inc/types" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/../codal-nrf52/nrfx/hal" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/sdk_config" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK_mods" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/pwr_mgmt"-I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/sortlist" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/strerror" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/softdevice/common" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/crc32" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/bootloader/dfu" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/util" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/common" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/balloc" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/peer_manager" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/ringbuf" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/timer" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/log" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/ble/nrf_ble_gatt" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/bootloader" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/fstorage" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/experimental_section_vars" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/softdevice/s113/headers" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/mutex" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/delay" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/bootloader/ble_dfu" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/atomic_fifo" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/atomic" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/memobj" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-nrf5sdk/nRF5SDK/components/libraries/fds" -I"C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/li [... cut ...] al-microbit-v2/CMakeFiles/codal-microbit-v2.dir/source/NRF52LedMatrix.cpp.obj -c "C:/Users/T Daniel/PycharmProjects/microbit-v2-samples/libraries/codal-microbit-v2/source/NRF52LedMatrix.cpp"
AR10B2~1.EXE: error: Daniel/PycharmProjects/microbit-v2-samples/build/codal_extra_definitions.h: No such file or directory
AR10B2~1.EXE: error: Daniel/PycharmProjects/microbit-v2-samples/utils/cmake/toolchains/ARM_GCC: No such file or directory
ninja: build stopped: subcommand failed.
```

Sorry for the mess

microbit-carlos, 2022-03-18:
@TDanielI, are you still experiencing this issue?

microbit-carlos, 2022-05-23:
I'll close this due to lack of activity, but if you are still experiencing this problem please free feel to reopen this issue! 👍

closed 2022-05-23 by microbit-carlos