lancaster-university/microbit-v2-samples #35: Building on Ubuntu Linux

TDanielI, 2021-10-03:
After encountering problems in windows os, i tried to compile files in ubuntu linux instead and i managed to successfully compiled the main.cpp file (the hello world program). I tried to move the sample file out from the source file and copied the BlinkyTest.cpp and Test.h to the source file alone and tried to redo the same process. I got this error:
```
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/softfp/crt0.o: in function `_mainCRTStartup':
/build/newlib-CVVEyx/newlib-3.3.0/build/arm-none-eabi/thumb/v7e-m+fp/softfp/libgloss/arm/semihv2m/../../../../../../../../libgloss/arm/crt0.S:545: undefined reference to `main'
```
Did I do any wrong steps on this or how can I use the compiler properly?

microbit-carlos, 2021-10-05:
Hi @TDanielI,
The test files don't have a `main()` function to run, they contain test functions to be called inside the original `main.cpp` file.

TDanielI, 2021-10-05:
What syntax should I use to call those functions?

microbit-carlos, 2021-10-05:
Only a few take arguments: https://github.com/lancaster-university/microbit-v2-samples/blob/master/source/samples/Tests.h

You need to include that header file and then call the functions.

microbit-carlos, 2022-03-18:
Hi @TDanielI, were you able to get this working?