lancaster-university/microbit-v2-samples #43: Remove `NO_BLE` from codal.json if it's not used anywhere.

carlosperate, 2022-02-23:
Had a search through the micro:bit CODAL repos I couldn't find any usage of the `NO_BLE` define. If it's not used anywhere we should remove it from the codal.json file:
https://github.com/lancaster-university/microbit-v2-samples/blob/ac325c50b5b13ff0b3355ac8b495bcc1ba2c2853/codal.json#L10-L11

microbit-carlos, 2022-02-24:
There is also `SOFTDEVICE_PRESENT`, which looks like it might be already defined in target.json

https://github.com/lancaster-university/microbit-v2-samples/blob/ac325c50b5b13ff0b3355ac8b495bcc1ba2c2853/codal.ble.json#L10-L11

https://github.com/lancaster-university/codal-microbit-v2/blob/6e880bc3d10df1a585bb9046ee4ab02bdcab1f34/target.json?q=SOFTDEVICE_PRESENT#L48

Can setting `"SOFTDEVICE_PRESENT": 0` in the JSON file remove this definition from the build?
If not, we might want to either add that feature, to habilitate CODAL builds without a SoftDevice, or remove it from the target.ble.json example file.

As a side note, the SoftDevice might be used by CODAL for other things apart from BLE, for example the internal nRF52 temperature sensor could be read using the SD, so that could also be a dependency on it.

microbit-carlos, 2022-02-24:
And if possible we might also want to merge `DEVICE_BLE ` and `MICROBIT_BLE_ENABLED`? :
https://github.com/lancaster-university/microbit-v2-samples/blob/ac325c50b5b13ff0b3355ac8b495bcc1ba2c2853/codal.ble.json#L10-L13

microbit-carlos, 2023-01-18:
Removed in cb9278e2ddf4c827547470adbd4973ce1502ae4f.