lancaster-university/microbit-v2-samples #49: Could this be modified for V1 microbit?

keble6, 2022-07-05:
I got this working easily - but I did not succeed with the V1 offline tools!
More detail on that here: https://support.microbit.org/support/tickets/54936

I'd like to develop some V1 code (there are lots of them out there, especially in schools, and lots of example code) for measuring microbit VDD - [example](https://os.mbed.com/teams/microbit/code/microbit-battery-test//file/6c9081f485dd/main.cpp/)

microbit-carlos, 2022-09-27:
Code for V1 can still be compiled using Yotta and the V1 https://github.com/lancaster-university/microbit-samples repository.

Documentation about the offline toolchain can be found in: https://lancaster-university.github.io/microbit-docs/offline-toolchains/

microbit-carlos, 2023-01-03:
I'll close this as resolved, but feel free to reopen if you have any other questions!

closed 2023-01-03 by microbit-carlos