lancaster-university/microbit-v2-samples #50: CMake Error

hnwangkg-ezio, 2022-07-08:
Cloning into: https://github.com/lancaster-university/codal-microbit-v2
Cloning into 'codal-microbit-v2'...
Checking out branch: master
Already on 'master'
Your branch is up to date with 'origin/master'.
Set target: codal-microbit-v2 
Using target.json (dev version) 
Targeting codal-microbit-v2
CMake Error at /usr/share/cmake-3.10/Modules/CMakeDetermineSystem.cmake:100 (message):
  Could not find toolchain file: ./utils/cmake/toolchains/toolchain.cmake
Call Stack (most recent call first):
  CMakeLists.txt:104 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

microbit-carlos, 2022-08-10:
Hi @hnwangkg-ezio 

It looks like your system might be misconfigured.
Could you start from zero and copy and paste all the commands and their output?

If you run this on a terminal, what does it output?
```
cmake --version
```

```
arm-none-eabi-gcc --version
```

o92, 2022-12-25:
Encountered same issue. Came across this thread and installed https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-win32.exe?rev=29bb46cfa0434fbda93abb33c1d480e6&hash=B2C5AAE07841929A0D0BF460896D6E52

The problem got fixed. 

My cmake is already at version 3.25.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

microbit-carlos, 2023-01-03:
Thanks for the additional info @o92!
As mentioned, I believe the original issue is related to CMake not finding `arm-none-eabi-gcc` automatically. Ensuring you have an environment with the Arm toolchain correctly installed and accessible (available via `PATH`) should resolve the issue.
You can always reinstall the Arm toolchain using the latest release and that should help as well: https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

I'll close this as resolved, but @hnwangkg-ezio if you still have issues please feel free to reopen and we can look into those 👍

closed 2023-01-03 by microbit-carlos