lancaster-university/microbit-v2-samples #51: Help with accessing ADC registers

keble6, 2022-07-10:
I'm struggling with setting up pointers etc to access the V2 (nRF52833) ADC registers.

The V1 has an ADC example [here](https://os.mbed.com/teams/microbit/code/microbit-battery-test//file/6c9081f485dd/main.cpp/).

How can I find out the equivalent "defines" (such as NRF_ADC->CONFIG) for the V2?

microbit-carlos, 2022-08-10:
Hi @keble6,

The register names can be found in the nRF52833 datasheet:
- https://infocenter.nordicsemi.com/index.jsp?topic=%2Fstruct_nrf52%2Fstruct%2Fnrf52833.html

And you should be able to find the structure definitions in the nrfx library (which is included in CODAL via [codal-nrf52]):
- (https://github.com/lancaster-university/codal-nrf52/): https://github.com/NordicSemiconductor/nrfx

keble6, 2022-08-10:
Thanks Carlos!
Rob

***@***.***>

microbit-carlos, 2022-09-27:
If that helped I think I'll close this as resolved, but let us know if you have any other questions.