lancaster-university/microbit-v2-samples #70: Python scripts no longer Python 2 compatible

microbit-carlos, 2023-06-15:
With the addition of the `parallelism` flag we introduced the use of f-strings: https://github.com/lancaster-university/microbit-v2-samples/commit/c854391ef5d7c9505fbe8c43cdeb3e68dcb380a3

I think it's quite possible MakeCode is running the build.py script with `python`, which will likely point to Python 2 in the docker images used:

https://github.com/search?q=repo%3Amicrosoft%2Fpxt%20build.py&type=code
[image]

microbit-carlos, 2023-06-15:
Okay, so the images at least do have `python3` installed, but `python` invokes Python 2.

```
$ docker run --rm pext/yotta:latest python --version
Python 2.7.6

$ docker run --rm pext/yotta:latest python3 --version
Python 3.4.3
```

```
$ docker run --rm pext/yotta:update-yotta3 python --version
Python 2.7.17

$ docker run --rm pext/yotta:update-yotta3 python3 --version
Python 3.6.9
```

microbit-carlos, 2023-06-29:
@JohnVidler MakeCode is still using the `pext/yotta:latest` docker image, which means `python3` is v3.4. We might need to remove the f-strings after all, as they were introduced in Python 3.6.

JohnVidler, 2023-07-04:
We can always roll back to a python 2 style string, if that's the safest?

JohnVidler, 2023-07-05:
Patched out the F-strings as of f4aaca3a9ff55dfd79c9ec41024dbbc0bec4de04 on master.

Added in a couple of new args for `--status` while I was at it, including `-v` compliance for increased verbosity, and `-n` to control the number of log lines. Specifying a lib name on a `--status` call now also only shows that library on its own.

Example run: `./build.py -svn 12 codal-core` - aka, run versbose, with 12 log liens for _only_ the `codal-core` lib.

All flag combinations are handled correctly.

Note: The current default parallelism is `10` which should be fine for most systems, but is possibly suboptimal for 'big' build systems (like Makecode's backend, perhaps?) which may want to bump this up via `-j [number]` or `--parallelism [number]` for better build performance.

Marking this as complete. @microbit-carlos can you re-open this please if it fails to build correctly on our upstreams (Makecode/Micropython, etc.)

closed 2023-07-05 by JohnVidler

microbit-carlos, 2023-07-05:
As soon as we get https://github.com/lancaster-university/codal-microbit-v2/issues/338 fixed we can test it in the codal-microbit-v2 repo as well:
- https://github.com/lancaster-university/codal-microbit-v2/issues/338