lancaster-university/microbit-v2-samples #97: Custom instantiated NRF52Serial object does not fire CODAL_SERIAL_EVT_DATA_RECEIVED events

muzkr, 2024-12-17:
Hi! I'm trying to get this simple echo program to work. The problem is that the CODAL_SERIAL_EVT_DATA_RECEIVED event is never fired. If I use the built-in uBit.serial object it works as expected. But I do need to instantiate another NRF52Serial object. Please help, thanks!
```cpp
#include "MicroBit.h"
#include "samples/Tests.h"

#define USE_UBIT_SERIAL 0

MicroBit uBit;

#if USE_UBIT_SERIAL
static NRF52Serial &serial = uBit.serial;
#else
static NRF52Serial serial(uBit.io.P13, uBit.io.P14, NRF_UARTE1);
#endif

static void on_rx(Event e)
{

    uint8_t buf[64];

    while (1)
    {
        int len = serial.read(buf, sizeof(buf), ASYNC);
        if (len > 0)
        {
            serial.send(buf, len, ASYNC);

            if (len == sizeof(buf))
            {
                continue;
            }
        }

        break;
    } // while
}

int main()
{

    uBit.init();

#if USE_UBIT_SERIAL
    serial.redirect(uBit.io.P13, uBit.io.P14);
#endif

    serial.send("hello!\n");

    uBit.messageBus.listen(DEVICE_ID_SERIAL, CODAL_SERIAL_EVT_DATA_RECEIVED, on_rx);

    uBit.sleep(100);

    while (1)
    {
        serial.send(".");
        uBit.sleep(1000);
    }

    microbit_panic(999);
}

```

muzkr, 2024-12-18:
I found out what the problem is. The newly instantiated NRF52Serial object is not RX ready (e.g., `initialiseRx()` is not called). A solution could be to call the `isReadable()` method, which will in turn call `initialiseRx()`. 

I'm going to leave this issue open, to bring it to the attention of micro:bit developers. Close it if you feel appropriate.

martinwork, 2024-12-18:
Hi @muzkr In the example, uBit.serial behaves differently because it is initialised by redirect.

muzkr, 2024-12-19:
Hi Martin @martinwork thanks for your response! Yes, it was the `redirect()` method that I was looking at earlier, and realized that the new object needed to be initialized with the RX.

closed 2024-12-20 by muzkr