lancaster-university/microbit-v2-samples #98: C++ extension: what's the proper way to specify enum values by macros?

muzkr, 2024-12-20:
Hi,

I'm developing a C++ extension that requires to define enums. If I specify the enum values to macro defined in a header file, `pxt build` fails.

Enum is defined as follows: (Full C++ code [here](https://github.com/imq-open/pxt-serial2/blob/master/serial2.cpp))

```cpp
enum EventBusSource
{
    //% blockIdentity="control.eventSourceId"
    SERIAL2_DEVICE_ID = IMQOPEN_NRF52SERIAL2_DEFAULT_DEVICE_ID,
}; 
```

`pxt build` error messages: 

```
error: Extension this:
   pxt-serial2/serial2.cpp(21): cannot determine value of     SERIAL2_DEVICE_ID = IMQOPEN_NRF52SERIAL2_DEFAULT_DEVICE_ID,
```
For now, I'm specifying the enum value as literals, but it would be better to use header files/macro definitions. What's the right way to do it?

martinwork, 2024-12-22:
@muzkr That question might be better asked in the MakeCode micro:bit forum (https://forum.makecode.com/) or repo https://github.com/microsoft/pxt-microbit.

I searched for the error message...
https://github.com/search?q=repo%3Amicrosoft%2Fpxt%20cannot%20determine%20value%20of&type=code
Is it looking for an integer or another enum rather than a macro?

Would it be better for the extension to include handlers for it's events,  maybe with friendlier names, rather than adding IDs to control.onEvent? I'm not sure...

By the way, I think namespace _serial2 is not necessary. and It can be changed to serial2 like this:
- delete the shims.d.ts file and remove it from pxt.json
- change all _serial2 to serial2
- delete the jsdoc comments from the cpp file, or change /** to /*, so they are just comments

Why duplicate NRF52Serial?

Check the requirements for the copied code:
https://github.com/lancaster-university/codal-nrf52/blob/master/LICENSE#L12
https://github.com/microsoft/pxt-microbit/blob/master/LICENSE.txt#L16

Is 70 a suitable event source ID to use? 
@microbit-carlos ?

muzkr, 2024-12-23:
My good Martin @martinwork came to my rescue once again, thank you very much!

I agree, it is not the right place here for this issue.

Ok, pxt does not support macro expansion for enum values. New magic recipe learned!

Friendly custom event registration functions: Good idea but maybe later. Also I prefer reusing existing facilities. 

Namespace _serial2 change: this is NO GOOD. Local pxt built fine, however, after loading it into the web editor, bad things  happen. The simulator disappears, and most of the files' contents display blank. Maybe you'd like to try it yourself?

License updated & third party software licensing obligations fulfilled.

martinwork, 2024-12-23:
@muzkr Yes, I only recently found this recipe, so I tested it first! I think you made it work now right? Did I miss something in the list of changes? 

Since extension ts shims must have a body, the generated shims.d.ts is not helpful (maybe temporarily, as a template?), and with no jsdoc comments or metadata in the cpp file, except //% on functions, shims.d.ts is not required by MakeCode or local build. Previously, I have marked the namespace with //% which requires a mostly empty shims.d.ts, but I realised that's not necessary.

muzkr, 2024-12-23:
@martinwork Yes it works now. And no, nothing was missed. I followed your 3-step recipe exactly. I think your reasoning about shims makes great sense. Earlier I have been thinking that the generated shims are **declarations** and it's the extension author's responsibilities to **implement** them, and soon I realised that I was wrong and pxt does not work that way. Provided simulator stub implementations have to be of **different** signatures to those of the generated declarations. Thats why I used the _serial2 namespace ealier.

muzkr, 2024-12-27:
Well the serial2 extension has been used by another extension of mine (imq-open/pxt-qbuild, essentially it turns Microbit into a lego sensor) and seems work reliably. I am very satisfied with the results. These are my first Micro:bit extensions and I am so impressed by the pxt toolkit and the codal runtime! Thank you @martinwork for your invaluable help! Issue closed.

closed 2024-12-27 by muzkr